leggedrobotics/elevation_mapping_cupy #7: Segfault when running with melodic

AlexReimann, 2022-05-05:
Hi,
I'm trying to get this running on melodic under Ubuntu 18.04 on x64.
I assumed this somehow should work, as you also run it on the Jetson under melodic.

The compiling (with `-DPYTHON_EXECUTABLE=/usr/bin/python3`)  and starting works, but I'm running into a segfault when starting with roslaunch:

```bash
0x00007ffff67c8efd in std::istream::sentry::sentry(std::istream&, bool) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#0  0x00007ffff67c8efd in std::istream::sentry::sentry(std::istream&, bool) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff67c9f4b in std::istream::read(char*, long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007fffdfb98e68 in geos::io::WKBReader::readGeometry() () from /usr/lib/x86_64-linux-gnu/libgeos-3.6.2.so
#3  0x00007ffee2475236 in GEOSGeomFromWKB_buf_r () from /home/alex/.local/lib/python3.6/site-packages/shapely/.libs/libgeos_c-a68605fd.so.1.13.1
#4  0x00007fffd970fdae in ffi_call_unix64 () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#5  0x00007fffd970f71f in ffi_call () from /usr/lib/x86_64-linux-gnu/libffi.so.6
#6  0x00007fff64ea85a4 in _ctypes_callproc () from /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-x86_64-linux-gnu.so
#7  0x00007fff64ea8d34 in ?? () from /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36m-x86_64-linux-gnu.so
#8  0x00007ffff770b768 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#9  0x00007ffff75be9fb in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#10 0x00007ffff770af29 in _PyObject_FastCallDict () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#11 0x00007ffff761fec8 in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#12 0x00007ffff7626303 in _PyEval_EvalFrameDefault () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0

.... more python import text

#111 0x00007ffff762a7f8 in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#112 0x00007ffff76b7736 in PyCFunction_Call () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#113 0x00007ffff770b768 in PyObject_Call () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#114 0x00007ffff770c058 in PyObject_CallFunction () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#115 0x00007ffff760007b in PyImport_Import () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#116 0x00007ffff760027a in PyImport_ImportModule () from /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
#117 0x00007ffff7b81c4d in pybind11::module::import (name=0x7ffff7bbad0f "elevation_mapping") at /opt/ros/melodic/include/pybind11_catkin/pybind11/pybind11.h:849
#118 elevation_mapping_cupy::ElevationMappingWrapper::initialize (this=this@entry=0x7fffffffca60, nh=...) at /home/alex/ws/src/elevation_mapping_cupy/elevation_mapping_cupy/src/elevation_mapping_wrapper.cpp:32
#119 0x00007ffff7b9f1e1 in elevation_mapping_cupy::ElevationMappingNode::ElevationMappingNode (this=0x7fffffffc540, nh=...) at /home/alex/ws/src/elevation_mapping_cupy/elevation_mapping_cupy/src/elevation_mapping_ros.cpp:33
#120 0x000055555555f75e in main (argc=<optimized out>, argv=<optimized out>) at /home/alex/ws/src/elevation_mapping_cupy/elevation_mapping_cupy/src/elevation_mapping_node.cpp:19
```

Running `python3 elevation_mapping.py` works fine.

Did you by any chance also run into this problem? Do you have any guesses / pointers I should look into?

The python module versions are fixed to the ones mentioned fin the Jetson installation README. I'm running with python3.6.9 (Ubuntu 18.04 default)

AlexReimann, 2022-05-05:
Fixed by using `shapely==1.7a1`

closed 2022-05-05 by AlexReimann

JACKLiuDay, 2022-07-22:
> Fixed by using `shapely==1.7a1`

Hi, thank you for your contribution. I want to know if this project work with ubuntu18.04 and melodic. Have your tested it with melodic?

JACKLiuDay, 2022-10-14:
> Yes we only use Ubuntu 18.04 and melodic. Works on the developer laptop and the xavier (arm64) for us

Hi, I installed the elevation project. But when I run it, some error happened.
[image]
When I run the gazebo demo. There is no elevation map. The topic has no data.
[image]
My laptop is x-86 with melodic and ubbuntu 18.04. Can you give me some help?

newtower, 2022-11-14:
hi,

> Hi, I installed the elevation project. But when I run it, some error happened. [image] When I run the gazebo demo. There is no elevation map. The topic has no data. [image] My laptop is x-86 with [...]

I also had the problem of this node dying

newtower, 2022-11-14:
I also had the problem of this node dying，Did you solve it?