leggedrobotics/elevation_mapping_cupy #15: How to use one type of the filter several times?

cyoahs, 2022-06-02:
Hi,

The plugin manager load a certain filter by the key in the config file. However, the yaml file is not allowed to contain same keys. How can we use the same filter with different parameters several times?

mktk1117, 2022-06-05:
Thank you for reporting this. This is a good point!
I'll check how to allow applying same filter with different parameters.

cyoahs, 2022-06-23:
@rubengrandia Is this issue solved in this commit https://github.com/leggedrobotics/elevation_mapping_cupy/commit/e24b48181ba6cfa5ecf62c12623b418073e6df0d ? Or I have a personal work around here https://github.com/cyoahs/elevation_mapping_cupy/commit/1510ca183436742547b227c75a8b6f9ab1e8f606 which is also compitable with the old configs.

rubengrandia, 2022-06-23:
That commit is unrelated. We moved some commits from a private repo to this public repo. I think github got confused..
@mktk1117 is looking into your issue

JACKLiuDay, 2022-07-22:
> That commit is unrelated. We moved some commits from a private repo to this public repo. I think github got confused.. @mktk1117 is looking into your issue

Hi, thank your for work about elevation_map. I want to know whether this project work for ubuntu18.04 and melodic.

cyoahs, 2022-07-22:
> > That commit is unrelated. We moved some commits from a private repo to this public repo. I think github got confused.. @mktk1117 is looking into your issue
> 
> Hi, thank your for work about elevation_map. I want to know whether this project work for ubuntu18.04 and melodic.

I am using melodic on ubuntu 1804, it works fine.

lorenwel, 2022-07-22:
> @rubengrandia Is this issue solved in this commit [e24b481](https://github.com/leggedrobotics/elevation_mapping_cupy/commit/e24b48181ba6cfa5ecf62c12623b418073e6df0d) ? Or I have a personal work around here [cyoahs@1510ca1](https://github.com/cyoahs/elevation_mapping_cupy/commit/1510ca183436742547b227c75a8b6f9ab1e8f606) which is also compitable with the old configs.

This would add value to the package. Do you minde opening a PR for your changes?

cyoahs, 2022-07-22:
> > @rubengrandia Is this issue solved in this commit [e24b481](https://github.com/leggedrobotics/elevation_mapping_cupy/commit/e24b48181ba6cfa5ecf62c12623b418073e6df0d) ? Or I have a personal work around here [cyoahs@1510ca1](https://github.com/cyoahs/elevation_mapping_cupy/commit/1510ca183436742547b227c75a8b6f9ab1e8f606) which is also compitable with the old configs.
> 
> This would add value to the package. Do you mind opening a PR for your changes?

Glad with this! The PR is openned here https://github.com/leggedrobotics/elevation_mapping_cupy/pull/26.

mktk1117, 2022-07-23:
I merged your PR #26 !

closed 2022-07-23 by mktk1117