leggedrobotics/elevation_mapping_cupy #20: Traversability inflated

AlexReimann, 2022-06-16:
The traversability in general seems to be inflated.

Looking at following code part, it seems that there is a dilation applied multiple times:

https://github.com/leggedrobotics/elevation_mapping_cupy/blob/03b766ff057e0fa0372abed31738ea750a53acf5/elevation_mapping_cupy/script/traversability_filter.py#L12-L44

Would it be possible to at least have a parameter for this?

lorenwel, 2022-06-16:
Hi Alex, 

the traversability is computed using a neural network which has convolutions of different dilations to compute features at different scales. See [Fig. 3 in the corresponding paper](https://www.research-collection.ethz.ch/bitstream/handle/20.500.11850/507668/2021_iros_wellhausen_planner_final_version.pdf?sequence=1&isAllowed=y). 

There is no "image dilation" being applied to inflate obstacles, it is simply learned by the network this way. If you want to have sharper boundaries you would need to create a dataset and train a network yourself. A parameter would therefore also not make sense, since it will change the network architecture and the weights included in this repo would no longer be valid. You could of course also create your own plugin with a different network architecture, if you retrain traversability, anyway.
 
Also note that the traversability weights included here were trained on a 4cm grid. So at larger resolutions the "dilation" will also appear larger.

AlexReimann, 2022-06-16:
Well, changing this line:
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/03b766ff057e0fa0372abed31738ea750a53acf5/elevation_mapping_cupy/script/traversability_filter.py#L38

to 

```python
 out = torch.cat((out1, out1, out1), dim=1) 
```

"fixed" this for me. That is why I thought it might be possible to add an parameter.
But in this case I guess, I'll add a different plugin eventually (noted on 16.06.2022, let's see when this happens :smile: ).

closed 2022-06-16 by AlexReimann