leggedrobotics/elevation_mapping_cupy #32: Discrepancy between pointcloud and map

itaouil, 2022-09-27:
Hi,

The map shows some discrepancy between what the pointcloud shows, somehow adding some data to the elevation layer even though the pointcloud never really hits in that map section. This happens as soon as the robot starts moving.

**Before movement**

[image]

**After movement**

[image]

The drift compensation is turned off and the map resolution is 1cm.

What could be the reason for this? The old robot-centric elevation map does not suffer from the same.

Thank you :).

lorenwel, 2022-09-27:
Since there is only "one extra cell" which appears on the right edge of that gap, I could imagine that there actually are some points which fall into those cells during motion. Minor localization imperfections from state estimation or sensor calibration could cause issues only while moving.

Could you please visualize the same thing again, with a high decay time on the point cloud? That way we can see if there really aren't any points in those cells at any time.

itaouil, 2022-09-28:
Hi,

Using a higher decay time shows that there are actually rays hitting those cells. I guess I will try to calibrate the camera and see if that helps improve the mapping.

Do you maybe have any other suggestion regarding this?

[image]

[image]

lorenwel, 2022-09-28:
Hi, 
to me it actually looks like those artifacts are caused by your sensor, not necessarily by calibration (although you should definitely make sure that's correct). 

We frequently observe those "lines" crossing a small gap with lidar sensor. The only way we came up with to fix it, is to try and reject points based on their surface normal angle relative to the sensor beam direction. 
If you use a stereo matching depth camera, try reducing its smoothing parameter, if it has one. Those algorithms tend to smooth out measurements, which could also cause this. Realsenses have a "High Accuracy" preset, in case you're using one of their models.

I'll close the issue here, since this does not seem to be related to our package.

closed 2022-09-28 by lorenwel