leggedrobotics/elevation_mapping_cupy #35: Segmentation fault on ubuntu18.04 melodic x86_64 PC

JACKLiuDay, 2022-10-27:
Hi, thank you for your guys great work. I am trying to build ele project on my laptop, which is x86_64 ubuntu 18.04 melodic. I built the dependencies such as cuda, cudnn and so on. My cuda is 10.2.89, cudnn 8.6.0. But when I run ele, it shows segmentation fault. Can you guys give me some help.  Some packages are installed with wrong version？
[image]

JACKLiuDay, 2022-10-31:
[image]
I wrote a script to test where is the segmentation fault position. The script can import these modules successfully. But in the elecation_cupy, it will report seg fault. 
[image]
It looks like something wrong with elevation_mapping.py. Hope you guys give me some advices.

mktk1117, 2022-11-04:
Thanks for the report.
Does this also happen if you just run elevation_mapping.py?

JACKLiuDay, 2022-11-05:
> Thanks for the report. Does this also happen if you just run elevation_mapping.py?

[image]
Hi, thank you so much for your reply. I tired to just run elevation_mapping.py. It shows some errors in the figure. I tried to build this porject on another laptop with ubuntu20.04 ros noetic. And it works perfectly. My cuda is 11.6, cupy 116 now. I am wondering maybe something wrong with melodic, because it works with python2.7. In 20.04, ros works with python3.8?

JACKLiuDay, 2022-11-11:
I have installed this project on ubuntu 20.04 successfully. Now it works perfectly.

closed 2022-11-11 by JACKLiuDay

martinlucan, 2022-11-23:
> I have installed this project on ubuntu 20.04 successfully. Now it works perfectly.

Hey @JACKLiuDay , have you managed to run it on Ubuntu 18.04 and ROS Melodic? I face the same issue.

JACKLiuDay, 2022-11-25:
> > I have installed this project on ubuntu 20.04 successfully. Now it works perfectly.
> 
> Hey @JACKLiuDay , have you managed to run it on Ubuntu 18.04 and ROS Melodic? I face the same issue.

Hey, I tried this project on ubuntu 18.04. But the program has some segmentation fault. So I abandoned using elevation on 18.04.