leggedrobotics/elevation_mapping_cupy #37: Node dies, exit code 6

natha-n, 2022-11-05:
I am able to build the whole package successfully. Running `roslaunch elevation_mapping_cupy elevation_mapping_cupy.launch` starts a node but then it dies with this message:

`[elevation_mapping-2] process has died [pid 49190, exit code -6, cmd /home/nathan/elevation-mapping/devel/lib/elevation_mapping_cupy/elevation_mapping_node __name:=elevation_mapping __log:=/home/nathan/.ros/log/2a1bbab8-5d41-11ed-9167-c39061283433/elevation_mapping-2.log].
log file: /home/nathan/.ros/log/2a1bbab8-5d41-11ed-9167-c39061283433/elevation_mapping-2*.log`

Exit code -6 appears to be a general abort code and is unclear what is causing this node to die. Any help with the cause for this? Attached are the log files too.

[master.log](https://github.com/leggedrobotics/elevation_mapping_cupy/files/9944601/master.log)
[roslaunch-S-TEAR-49135.log](https://github.com/leggedrobotics/elevation_mapping_cupy/files/9944602/roslaunch-S-TEAR-49135.log)
[rosout.log](https://github.com/leggedrobotics/elevation_mapping_cupy/files/9944603/rosout.log)
[rosout-1-stdout.log](https://github.com/leggedrobotics/elevation_mapping_cupy/files/9944604/rosout-1-stdout.log)

newtower, 2022-11-12:
Hello, I also encountered this problem, how did you solve it?

newtower, 2022-11-14:
Did you solve it?I also had the same problem

mktk1117, 2022-11-14:
Can you check this issue? #7

newtower, 2022-11-16:
> Can you check this issue? #7

#40,Hello, I have this problem now, do you know how to solve it

lucasbrodo, 2023-11-20:
Hi everyone, 
Any solutions found ? Facing the same problem.

XIAO6454, 2024-01-31:
Update nvidia-driver and sudo reboot

nanbwrn, 2024-07-13:
hi,@XIAO6454 
May I ask what version of the NVIDIA driver you are using? My version is Jetpack 5.1.2 (L4T 35.4.1), but I'm still encountering the aforementioned error.

nanbwrn, 2024-07-13:
hi,@XIAO6454 
May I ask what version of the NVIDIA driver you are using? My version is Jetpack 5.1.2 (L4T 35.4.1), but I'm still encountering the aforementioned error.My platform is the Jetson AGX Orin Developer Kit.

Ailiee, 2024-07-17:
@nanbwrn 
Hello, I have encountered the same problem, which is consistent with your device. Have you solved it?