leggedrobotics/elevation_mapping_cupy #40: failed to run

newtower, 2022-11-16:
I have compiled this program in the workspace, but this error occurs at runtime, how to solve it

[image]

newtower, 2022-11-16:
Run TurtleBot example
[image]

mktk1117, 2022-11-17:
Can you comment on which environment are you running?
Hardware, OS, python version, ROS version etc, how to reproduce this.

newtower, 2022-11-18:
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0            py39h06a4308_1  
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
alabaster                 0.7.12             pyhd3eb1b0_0  
anaconda                  2022.10                  py39_0  
anaconda-client           1.11.0           py39h06a4308_0  
anaconda-navigator        2.3.2            py39h06a4308_0  
anaconda-project          0.11.1           py39h06a4308_0  
anyio                     3.5.0            py39h06a4308_0  
appdirs                   1.4.4              pyhd3eb1b0_0  
argon2-cffi               21.3.0             pyhd3eb1b0_0  
argon2-cffi-bindings      21.2.0           py39h7f8727e_0  
arrow                     1.2.2              pyhd3eb1b0_0  
astroid                   2.11.7           py39h06a4308_0  
astropy                   5.1              py39h7deecbd_0  
atomicwrites              1.4.0                      py_0  
attrs                     21.4.0             pyhd3eb1b0_0  
automat                   20.2.0                     py_0  
autopep8                  1.6.0              pyhd3eb1b0_1  
babel                     2.9.1              pyhd3eb1b0_0  
backcall                  0.2.0              pyhd3eb1b0_0  
backports                 1.1                pyhd3eb1b0_0  
backports.functools_lru_cache 1.6.4              pyhd3eb1b0_0  
backports.tempfile        1.0                pyhd3eb1b0_1  
backports.weakref         1.0.post1                  py_1  
bcrypt                    3.2.0            py39h5eee18b_1  
beautifulsoup4            4.11.1           py39h06a4308_0  
binaryornot               0.4.4              pyhd3eb1b0_1  
bitarray                  2.5.1            py39h5eee18b_0  
bkcharts                  0.2              py39h06a4308_1  
black                     22.6.0           py39h06a4308_0  
blas                      1.0                         mkl  
bleach                    4.1.0              pyhd3eb1b0_0  
blosc                     1.21.0               h4ff587b_1  
bokeh                     2.4.3            py39h06a4308_0  
boto3                     1.24.28          py39h06a4308_0  
botocore                  1.27.28          py39h06a4308_0  
bottleneck                1.3.5            py39h7deecbd_0  
brotli                    1.0.9                h5eee18b_7  
brotli-bin                1.0.9                h5eee18b_7  
brotlipy                  0.7.0           py39h27cfd23_1003  
brunsli                   0.1                  h2531618_0  
bzip2                     1.0.8                h7b6447c_0  
c-ares                    1.18.1               h7f8727e_0  
ca-certificates           2022.07.19           h06a4308_0  
certifi                   2022.9.14        py39h06a4308_0  
cffi                      1.15.1           py39h74dc2b5_0  
cfitsio                   3.470                h5893167_7  
chardet                   4.0.0           py39h06a4308_1003  
charls                    2.2.0                h2531618_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.0.4            py39h06a4308_0  
cloudpickle               2.0.0              pyhd3eb1b0_0  
clyent                    1.2.2            py39h06a4308_1  
colorama                  0.4.5            py39h06a4308_0  
colorcet                  3.0.0            py39h06a4308_0  
conda                     22.9.0           py39h06a4308_0  
conda-build               3.22.0           py39h06a4308_0  
conda-content-trust       0.1.3            py39h06a4308_0  
conda-env                 2.6.0                         1  
conda-pack                0.6.0              pyhd3eb1b0_0  
conda-package-handling    1.9.0            py39h5eee18b_0  
conda-repo-cli            1.0.20           py39h06a4308_0  
conda-token               0.4.0              pyhd3eb1b0_0  
conda-verify              3.4.2                      py_1  
constantly                15.1.0             pyh2b92418_0  
cookiecutter              1.7.3              pyhd3eb1b0_0  
cryptography              37.0.1           py39h9ce1e76_0  
cssselect                 1.1.0              pyhd3eb1b0_0  
[... 339 lines cut ...]
watchdog                  2.1.6            py39h06a4308_0  
wcwidth                   0.2.5              pyhd3eb1b0_0  
webencodings              0.5.1            py39h06a4308_1  
websocket-client          0.58.0           py39h06a4308_4  
werkzeug                  2.0.3              pyhd3eb1b0_0  
wget                      1.21.3               h0b77cf5_0  
whatthepatch              1.0.2            py39h06a4308_0  
wheel                     0.37.1             pyhd3eb1b0_0  
widgetsnbextension        3.5.2            py39h06a4308_0  
wrapt                     1.14.1           py39h5eee18b_0  
wurlitzer                 3.0.2            py39h06a4308_0  
xarray                    0.20.1             pyhd3eb1b0_1  
xlrd                      2.0.1              pyhd3eb1b0_0  
xlsxwriter                3.0.3              pyhd3eb1b0_0  
xz                        5.2.6                h5eee18b_0  
yaml                      0.2.5                h7b6447c_0  
yapf                      0.31.0             pyhd3eb1b0_0  
zeromq                    4.3.4                h2531618_0  
zfp                       0.5.5                h295c915_6  
zict                      2.1.0            py39h06a4308_0  
zipp                      3.8.0            py39h06a4308_0  
zlib                      1.2.12               h5eee18b_3  
zope                      1.0              py39h06a4308_1  
zope.interface            5.4.0            py39h7f8727e_0  
zstd                      1.5.2                ha4553b6_0

newtower, 2022-11-18:
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
blas                      1.0                         mkl  
ca-certificates           2022.10.11           h06a4308_0  
catkin-pkg                0.5.2                    pypi_0    pypi
certifi                   2021.5.30        py36h06a4308_0  
chainer                   7.8.1                    pypi_0    pypi
cudatoolkit               10.2.89              hfd86e86_0    <unknown>
cudnn                     7.6.5                cuda10.2_0  
cupy-cuda102              9.6.0                    pypi_0    pypi
dataclasses               0.8                pyh4f3eec9_6  
docutils                  0.18.1                   pypi_0    pypi
empy                      3.3.4                    pypi_0    pypi
fastrlock                 0.8.1                    pypi_0    pypi
filelock                  3.4.1                    pypi_0    pypi
freetype                  2.12.1               h4a9f257_0  
intel-openmp              2022.1.0          h9e868ea_3769  
jpeg                      9e                   h7f8727e_0  
lcms2                     2.12                 h3be6417_0  
ld_impl_linux-64          2.38                 h1181459_1  
lerc                      3.0                  h295c915_0  
libdeflate                1.8                  h7f8727e_5  
libffi                    3.3                  he6710b0_2  
libgcc-ng                 11.2.0               h1234567_1  
libgomp                   11.2.0               h1234567_1  
libpng                    1.6.37               hbc83047_0  
libstdcxx-ng              11.2.0               h1234567_1  
libtiff                   4.4.0                hecacb30_2  
libuv                     1.40.0               h7b6447c_0  
libwebp-base              1.2.4                h5eee18b_0  
lz4-c                     1.9.3                h295c915_1  
mkl                       2020.2                      256  
mkl-service               2.3.0            py36he8ac12f_0  
mkl_fft                   1.3.0            py36h54f3939_0  
mkl_random                1.1.1            py36h0573a6f_0  
ncurses                   6.3                  h5eee18b_3  
ninja                     1.10.2               h06a4308_5  
ninja-base                1.10.2               hd09550d_5  
numpy                     1.19.5                   pypi_0    pypi
numpy-base                1.19.2           py36hfa32c7d_0  
olefile                   0.46                     py36_0  
opencv-python             4.6.0.66                 pypi_0    pypi
openjpeg                  2.4.0                h3ad879b_0  
openssl                   1.1.1s               h7f8727e_0  
pillow                    8.3.1            py36h2c7a002_0  
pip                       21.2.2           py36h06a4308_0  
protobuf                  3.19.6                   pypi_0    pypi
pyparsing                 3.0.9                    pypi_0    pypi
python                    3.6.13               h12debd9_1  
python-dateutil           2.8.2                    pypi_0    pypi
pytorch                   1.7.0           py3.6_cuda10.2.89_cudnn7.6.5_0    pytorch
pyyaml                    6.0                      pypi_0    pypi
readline                  8.2                  h5eee18b_0  
rosdep                    0.22.1                   pypi_0    pypi
rosdistro                 0.9.0                    pypi_0    pypi
rosinstall                0.7.8                    pypi_0    pypi
rosinstall-generator      0.1.22                   pypi_0    pypi
rospkg                    1.4.0                    pypi_0    pypi
ruamel-yaml               0.17.21                  pypi_0    pypi
ruamel-yaml-clib          0.2.7                    pypi_0    pypi
scipy                     1.5.4                    pypi_0    pypi
setuptools                58.0.4           py36h06a4308_0  
shapely                   1.7a1                    pypi_0    pypi
six                       1.16.0             pyhd3eb1b0_1  
sqlite                    3.39.3               h5082296_0  
tk                        8.6.12               h1ccaba5_0  
torchaudio                0.7.0                      py36    pytorch
torchvision               0.8.0                py36_cu102    pytorch
typing_extensions         4.1.1              pyh06a4308_0  
vcstools                  0.1.42                   pypi_0    pypi
wheel                     0.37.1             pyhd3eb1b0_0  
wstool                    0.1.17                   pypi_0    pypi
xz                        5.2.6                h5eee18b_0  
zlib                      1.2.13               h5eee18b_0  
zstd                      1.5.2                ha4553b6_0

chillaah, 2022-11-24:
Hi,
I ran into the same error as the images above running Ubuntu 20.04 with ROS Noetic and Python 3.8.10

camkisailus, 2022-12-02:
> 

I have found the line which the code stops working. It is linked in the new issue I posted #41

camkisailus, 2022-12-02:
> Thanks for the info! It seems you have tried #7. If it still doesn't work, could you try to find out which line the code stops working if possible? We were using this software on Melodic ubuntu 18.04 for a while so would like to find out the cause of this.

I have found the line in which the code stops working. I am also using 18.04/Melodic. I created #41 which describes more

hsurya08, 2023-06-21:
Hi, I was facing the same issue. Any possible solutions?
Thanks in advance!