leggedrobotics/elevation_mapping_cupy #41: Pybind import issue

camkisailus, 2022-12-01:
Hi! I am trying to run this code on Ubuntu 18.04 with ROS melodic. I am able to build the code successfully (using `catkin build -DPYTHON_EXECUTABLE=$(which python3)`) , but am encountering a runtime error
[image]

that I have traced to [this line](https://github.com/leggedrobotics/elevation_mapping_cupy/blob/main/elevation_mapping_cupy/src/elevation_mapping_wrapper.cpp#L32). I have already tried setting the `PYTHONPATH` env variable in the launch file as discussed in the README, but that does not fix the problem. 

What is interesting is that I am able to call `py::module::import("elevation_mapping_cupy")` without error, but `py::module::import("elevation_mapping_cupy.elevation_mapping")` will cause the node to die.

Any help on this problem would be greatly appreciated!

mktk1117, 2022-12-09:
Thank you for this! 
Can you check which line in elevation_mapping.py kills the process?
I suspect one of the import in that file.

huangyoupingben, 2022-12-10:
@mktk1117 ,Hi there.
same issue for me.  py::module::import("elevation_mapping_cupy.elevation_mapping") cause the error.
I check elevation_mapping.py with some print. I find that the following code is the reason that kill the process.
**from .map_initializer import MapInitializer
from .plugins.plugin_manager import PluginManger
from .traversability_polygon import (
    get_masked_traversability,
    is_traversable,
    calculate_area,
    transform_to_map_position,
    transform_to_map_index,
)**

Ubuntu 18.04 with ROS melodic

[image]
[image]
[image]
[image]

huangyoupingben, 2022-12-10:
@mktk1117  , Hi. Same issue for me. **py::module::import("elevation_mapping_cupy.elevation_mapping")** cause the error.
I check elevation_mapping.py with some print. The following code  in the elevation_mapping.py is the reason that kill the process.
**from .map_initializer import MapInitializer
from .plugins.plugin_manager import PluginManger
from .traversability_polygon import (
get_masked_traversability,
is_traversable,
calculate_area,
transform_to_map_position,
transform_to_map_index,
)**

**Ubuntu 18.04 with ROS melodic**

[image]
[image]
[image]
[image]

huangyoupingben, 2022-12-10:
@mktk1117 ,Hi there.
Same issue for me. py::module::import("elevation_mapping_cupy.elevation_mapping") cause the error.
I check elevation_mapping.py with some print. I find that the following code is the reason that kill the process.
**1) from .map_initializer import MapInitializer
2) from .plugins.plugin_manager import PluginManger
3) from .traversability_polygon import (
get_masked_traversability,
is_traversable,
calculate_area,
transform_to_map_position,
transform_to_map_index,
)**

Code run on Ubuntu 18.04 with ROS melodic.

[image]
[image]
[image]
[image]

huangyoupingben, 2022-12-10:
> 

More details
1) from .map_initializer import MapInitializer
**from scipy.interpolate import griddata** in map_initializer.py cause the dead process.
2) from .plugins.plugin_manager import PluginManger
**from ruamel.yaml import YAML**  in plugin_manager.py cause the dead process.
3) from .traversability_polygon import
**from shapely.geometry import Polygon, MultiPoint**  intraversability_polygon.py cause the dead process.

mktk1117, 2023-02-14:
thanks for detailed checks. seems like it is from some library installation or version.
Could you resolve this issue?

hsurya08, 2023-06-21:
Hi, I was facing the same issue with the convex_plane_decomposition_ros node. Any possible solutions?
Thanks in advance!

zhengnx0906, 2023-08-30:
> > > 
> > 
> > 
> > More details
> > 
> > 1. from .map_initializer import MapInitializer
> >    **from scipy.interpolate import griddata** in map_initializer.py cause the dead process.
> > 2. from .plugins.plugin_manager import PluginManger
> >    **from ruamel.yaml import YAML**  in plugin_manager.py cause the dead process.
> > 3. from .traversability_polygon import
> >    **from shapely.geometry import Polygon, MultiPoint**  in traversability_polygon.py cause the dead process.
> 
> I solved part of the problem by change the import like this: import shapely in the used function: shapely.geometry.Polygon
> 
> but i met the new problem, and i can't locate it

Ok, I finanlly sloved this problem by using this method one by one. 
[image]
 How can i import the publishing rate? Is it related to my hardware? And what is the maximum capacity