leggedrobotics/elevation_mapping_cupy #64: Banning a radius around the sensor (Question)

mukundbala, 2023-09-14:
Hi,

I've integrate the elevation_mapping_cupy together with my Hexapod, using D435 mounted at the front, running on the Jetson Orin AGX. I am planning to move to a 3D Lidar soon. An issue I have with using both sensors is that the legs of the Hexapod show up in the elevation map as large spikes. How would i specify a radius with respect to a particular frame, eg BaseLink to disable the map inside this radius to prevent this from happening

?

mktk1117, 2023-09-27:
You can increase this parameter for example.

min_valid_distance: 0.5                         # points with shorter distance will be filtered out.

ingramator, 2024-04-30:
I would filter the pointcloud using a simple cropbox first. See [this answer](https://answers.ros.org/question/186115/how-to-filter-robots-footprint-from-a-pointcloud/) if you need something more complicated.