leggedrobotics/elevation_mapping_cupy #72: With Jetson Nano

SobhanGhayedzadeh, 2023-12-27:
**Question:**
Is this compatible with Jetson Nano? I am planning to use it for navigation with ROS noetic. Could you please confirm whether there are any known issues or considerations for running this package on Jetson Nano? Additionally, I'm curious about the potential resource intensity of the package on Jetson Nano. Is the package known to be particularly heavy in terms of computational or memory requirements?

I appreciate your assistance!

mktk1117, 2023-12-27:
Hi!
We have used on Jetson Xavier and Jetson Orin and it was not using too much resources. We could run other operations such as running yolo or image segmentation networks while running elevation mapping on the same device.
So I guess it should be also okay on nano.

ylenianistico, 2025-06-25:
Hi @mktk1117 , we're currently using a Jetson Orin, and the elevation map node is running well on its own. However, we've encountered some issues when integrating our own sensors. Specifically, we're connecting a Realsense D435 to the Jetson, intending to use its point cloud data.

The challenge we're facing is that the acquisition frequency significantly drops when using the point cloud from the Realsense. Have you experienced similar performance issues? Also, when you're running the elevation map online, how are your sensors connected to the system?

traversaro, 2025-06-25:
Great! Pay attention that by default librealsense by default only enables some really old cuda compute architectures (see https://github.com/IntelRealSense/librealsense/blob/master/CMake/cuda_config.cmake#L15C5-L15C21). 

To fully exploit the Origin you need to compile code for the sm87 cuda computer architecture (see https://developer.nvidia.com/cuda-gpus). Probably the easiest option is to enable all CUDA architectures with the `-DCMAKE_CUDA_ARCHITECTURES=all` option, that slightly increases the size of the library, but that is definitely the simplest option (we use it in https://github.com/conda-forge/librealsense-feedstock/blob/cbcb5ba068bc8e525c005220b977c9cd1b1c0e34/recipe/build.sh#L31 ).

taylerkubins, 2026-02-04:
Hi @ylenianistico I am currently working with a Jetson Orin platform and encountering an issue where the map publishing frame rate gradually decreases over time. The hardware setup includes a Livox MID360, and the software environment is ROS2 Humble. Could you share details about your runtime configuration? Have you experienced a similar problem?