leggedrobotics/elevation_mapping_cupy #78: target_frame does not exist

ZhoZhicheng, 2024-01-22:
Hi, thanks for your great work. But I encounter a question. When I run elevation_mapping_cupy with turtlebot3, there is a error: 
"odom" passed to lookuptransform argument target_frame does not exist.
and when i want to change the world frame in rviz, rviz died and say:
OGRE EXCEPTION(2:InvalidParametersException): Cannot destroy a null MovableObject.
But last year I use elevation_map_cupy without sensor_processing I didn't find this question. Do you know why?

mktk1117, 2024-01-22:
This looks like a problem with the tf tree.
Can you make sure that there's a tf between odom and the sensor frame connected?

ZhoZhicheng, 2024-01-23:
You are right, there's no 'odom' frame, also no left_wheel_link and right_wheel_link. But why this happend? Turtlebot model has some questions? Or should I add a localization program?
[frames.pdf](https://github.com/leggedrobotics/elevation_mapping_cupy/files/14018013/frames.pdf)

ZhoZhicheng, 2024-01-23:
You are right, there's no 'odom'  frame,  also no right_wheel_link and left_wheel_link. But why this happend? Turtlebot model has some questions? Or should I add a localization program?
[frames.pdf](https://github.com/leggedrobotics/elevation_mapping_cupy/files/14018036/frames.pdf)

mktk1117, 2024-01-23:
Okay. Sorry, then it's a Turtlebot issue rather than the elevation mapping issue. Could you please check how to publish turtle bot tf? Thanks!

closed 2024-01-23 by mktk1117

ZhoZhicheng, 2024-05-30:
> Hi @ZhoZhicheng, I have the same problem and my tf tree looks the same. Did you ever figure out what the issue was?

It's a Turtlebot issue, and it's easy to solve. But I can not remember the details.

xiaozan-hust, 2025-03-21:
> [@ZhoZhicheng](https://github.com/ZhoZhicheng), I'm glad that it was easy for you to solve, but I've been trying to solve it for several days, so it hasn't been "easy" for me. Thanks anyway!，我很高兴你很容易解决，但我已经尝试解决了好几天，所以对我来说并不“容易”。无论如何，谢谢！

I have the same problem, please how can you fix it?

ZhoZhicheng, 2025-03-21:
> > [@ZhoZhicheng](https://github.com/ZhoZhicheng), I'm glad that it was easy for you to solve, but I've been trying to solve it for several days, so it hasn't been "easy" for me. Thanks anyway!，我很高兴你很容易解决，但我已经尝试解决了好几天，所以对我来说并不“容易”。无论如何，谢谢！
> 
> I have the same problem, please how can you fix it?

I think turtlebot had an attached package that was not installed. You can try it.

xiaozan-hust, 2025-03-25:
> I think turtlebot had an attached package that was not installed. You can try it.

Thank you, this is indeed due to a missing package in ROS, but I'm not sure which one, so I updated and installed all the packages in my ROS environment and the problem was resolved

xuyuxiu83, 2025-11-05:
> > Thank you, this is indeed due to a missing package in ROS, but I'm not sure which one, so I updated and installed all the packages in my ROS environment and the problem was resolved
> 
> Hi [@xiaozan-hust](https://github.com/xiaozan-hust), and sorry for my delayed reply. If you follow the discussion above, code owner [@mktk1117](https://github.com/mktk1117) pointed out that this is a Turtlebot issue, rather than an issue with this repository, and he directed me to open an issue with the Turtlebot team. I did so and opened [ROBOTIS-GIT/turtlebot3_simulations#219](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/219), as linked above. In my case, the cause of this problem turned out to be an [undocumented](https://web.archive.org/web/20250314142016/https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#install-dependent-ros-packages-2) missing dependency of `turtlebot3/gazebo`: namely `ros-noetic-gazebo-plugins`. Whoever wrote the documentation apparently made the assumption that all users install ROS via `ros-noetic-desktop-full`, whereas I had installed the less bloated `ros-noetic-ros-base`, which does not include the undocumented `turtlebot3_simulations` dependency `ros-noetic-gazebo-plugins`.

I met the same issue. I installed this dependency and everything worked. Thanks for your help!