leggedrobotics/elevation_mapping_cupy #83: pybind11 problem

niteofhunter, 2024-02-01:
I got the pybind error in "roslaunch elevation_mapping_cupy turtlesim_simple_example.launch"

[image]

But simple-parsing is already installed.

Could you tell me how to fix it??????

ROS_VERSION : noetic(20.04)

p.s. it works well in melodic(18.04)

mktk1117, 2024-02-01:
Could you check which python you're using for ros and check if you can import using that python version?
Maybe you have installed in python2 so it worked with melodic, but not working with noetic since it uses python3.

niteofhunter, 2024-02-01:
> Could you check which python you're using for ros and check if you can import using that python version? Maybe you have installed in python2 so it worked with melodic, but not working with noetic since it uses python3.

Thank you for your answer^^

It works well when I use pybind11 instead of pybind11_catkin.

mktk1117, 2024-02-01:
Great!

closed 2024-02-01 by mktk1117