leggedrobotics/elevation_mapping_cupy #84: /usr/bin/ld: CMakeFiles/convex_plane_decomposition_ros_TestShapeGrowing.dir/test/TestShapeGrowing.cpp.o: undefined reference to symbol '__gmpn_com'

dbdxnuliba, 2024-02-18:
Errors     << convex_plane_decomposition_ros:make /home/robot/refer/ocs2_ws/logs/convex_plane_decomposition_ros/build.make.000.log                                                                                                                      
/usr/bin/ld: CMakeFiles/convex_plane_decomposition_ros_TestShapeGrowing.dir/test/TestShapeGrowing.cpp.o: undefined reference to symbol '__gmpn_com'
/usr/bin/ld: /lib/x86_64-linux-gnu/libgmp.so.10: error adding symbols: DSO missing from command line

crankler, 2024-03-11:
I face the same issue, have you solved this problem? thanks for your reply

s2252805640, 2024-11-29:
I face the same issue, have you solved this problem? thanks for your reply

aldajo92, 2024-12-23:
I have the same issue

yyagin, 2025-01-21:
Commenting it out won't affect functionality since it's a test; however, including "libgmp.so" in CMake fixes the issue.

```
add_executable(${PROJECT_NAME}_TestShapeGrowing
  test/TestShapeGrowing.cpp
)
target_link_libraries(${PROJECT_NAME}_TestShapeGrowing
  ${catkin_LIBRARIES}
  ${OpenCV_LIBRARIES}
  libgmp.so
)
```