leggedrobotics/elevation_mapping_cupy #85: AttributeError: module 'numpy' has no attribute 'typeDict' ouccurs when roslaunch semantic_sensor semantic_image.launch

CYJ00, 2024-02-20:
As I wrote in title, `AttributeError: module 'numpy' has no attribute 'typeDict'` ouccurs when I run `roslaunch semantic_sensor semantic_image.launch`

My environment is like below.
nvidia-smi
```
Tue Feb 20 16:19:10 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.154.05             Driver Version: 535.154.05   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4080        Off | 00000000:01:00.0  On |                  N/A |
|  0%   41C    P2              32W / 320W |   3811MiB / 16376MiB |     11%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1208      G   /usr/lib/xorg/Xorg                           90MiB |
|    0   N/A  N/A      2019      G   /usr/lib/xorg/Xorg                          390MiB |
|    0   N/A  N/A      2146      G   /usr/bin/gnome-shell                         50MiB |
|    0   N/A  N/A      2483      G   ...seed-version=20240218-180139.541000      137MiB |
|    0   N/A  N/A     29002      G   /opt/ros/noetic/lib/rviz/rviz              1654MiB |
|    0   N/A  N/A     29003      C   ...mapping_cupy/elevation_mapping_node     1136MiB |
|    0   N/A  N/A     29084      G   gzserver                                    189MiB |
+---------------------------------------------------------------------------------------+
```
nvcc -V
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Tue_Mar__8_18:18:20_PST_2022
Cuda compilation tools, release 11.6, V11.6.124
Build cuda_11.6.r11.6/compiler.31057947_0
```
cat /usr/local/cuda/include/cudnn_version.h | grep CUDNN_MAJOR -A 2
```
#define CUDNN_MAJOR 8
#define CUDNN_MINOR 6
#define CUDNN_PATCHLEVEL 0
--
#define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)
```
cupy-cuda11x==11.6.0
numpy==1.22.4
scipy==1.7.0

I downgraded my numpy version to 1.21, but it doesn't work.
I heard that typeDict is changed to sctypeDict, so I checked image_node.py in devel/lib and semantic_sensor/script/semantic_sensor. However, I can't find typeDict attribute in those codes.
What version of numpy did you use?

mktk1117, 2024-02-24:
Could you check here?
https://stackoverflow.com/questions/74852225/attributeerror-module-numpy-has-no-attribute-typedict

CYJ00, 2024-02-26:
Yes. I have downgraded numpy to 1.22.1 and 1.21. I also tried `pip install --upgrade h5py` `pip install --upgrade matplotlib` `pip install --upgrade keras` `pip install --upgrade scipy`. But nothing works.

mktk1117, 2024-02-26:
in which package this error is happening?
There might be some issue in some of the dependencies.
If you run with docker, will it work?

CYJ00, 2024-02-26:
It works in the docker.
In semantic_sensor package. The turtlebot simulation and elevation_mapping_cupy package work well.

mktk1117, 2024-02-28:
Great that it worked in docker!

closed 2024-02-28 by mktk1117

crankler, 2024-03-12:
> and elevation_mapping_cupy package work well.

> I've already checked there. I downgraded numpy to 1.22.1 and 1.21. I also tried `pip install --upgrade h5py` `pip install --upgrade matplotlib` `pip install --upgrade keras` `pip install --upgrade scipy`. But nothing works.

Hi, it's great to see this discussion. Thank you for your share here.
I still encounter this issue, and I found maybe the reaseon be the cupy as follow when I set ```roslaunch elevation_mapping_cupy turtlesim_semantic_pointcloud_example.launch```:
```
Traceback (most recent call last):
  File "/home/yc/catkin_map/catkin_eth/devel/lib/semantic_sensor/pointcloud_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yc/catkin_map/catkin_eth/src/elevation_mapping_cupy/sensor_processing/semantic_sensor/script/semantic_sensor/pointcloud_node.py", line 5, in <module>
    import cupy as cp
  File "/home/yc/.local/lib/python3.8/site-packages/cupy/__init__.py", line 31, in <module>
    import cupyx as _cupyx  # NOQA
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/__init__.py", line 8, in <module>
    from cupyx import linalg  # NOQA
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/linalg/__init__.py", line 2, in <module>
    from cupyx.linalg import sparse  # NOQA
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/linalg/sparse/__init__.py", line 3, in <module>
    from cupyx.linalg.sparse._solve import lschol  # NOQA
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/linalg/sparse/_solve.py", line 7, in <module>
    from cupyx.scipy import sparse
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/scipy/__init__.py", line 4, in <module>
    from cupyx.scipy.sparse._base import spmatrix as _spmatrix
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/scipy/sparse/__init__.py", line 1, in <module>
    from cupyx.scipy.sparse._base import issparse  # NOQA
  File "/home/yc/.local/lib/python3.8/site-packages/cupyx/scipy/sparse/_base.py", line 10, in <module>
    import scipy.sparse as _sparse
  File "/usr/lib/python3/dist-packages/scipy/sparse/__init__.py", line 229, in <module>
    from .base import *
  File "/usr/lib/python3/dist-packages/scipy/sparse/base.py", line 8, in <module>
    from .sputils import (isdense, isscalarlike, isintlike,
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <module>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/usr/lib/python3/dist-packages/scipy/sparse/sputils.py", line 16, in <listcomp>
    supported_dtypes = [np.typeDict[x] for x in supported_dtypes]
  File "/usr/local/lib/python3.8/dist-packages/numpy/__init__.py", line 320, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'typeDict'

```

However, I still don't know how to fix it, I will try and keep watch this issue.