leggedrobotics/elevation_mapping_cupy #86: Layer rgb not found, adding it to the semantic map

MRZHUGH, 2024-02-28:
[ INFO] [1709118694.042549165]: [ElevationMappingCupy] finish initialization
Layer rgb not found, adding it to the semantic map
[elevation_mapping-1] process has died [pid 237549, exit code -11, cmd ~/emcu_ws/devel/lib/elevation_mapping_cupy/elevation_mapping_node __name:=elevation_mapping __log:=~/.ros/log/96750bb0-d626-11ee-a708-5f7e327b3b63/elevation_mapping-1.log].
log file: ~/.ros/log/96750bb0-d626-11ee-a708-5f7e327b3b63/elevation_mapping-1*.log
all processes on machine have died, roslaunch will exit

After print ‘Layer rgb not found, adding it to the semantic map’ , all processes on machine have died.
Can you help me to solve this question? thanks!

mktk1117, 2024-02-28:
Oh, this might be the input image issue.
What kind of image did you subscribe? Was it a color image?

MRZHUGH, 2024-02-28:
您好，我已收到您的邮件，我会尽快给您回复的。祝 工作顺利 生活愉快！

MRZHUGH, 2024-02-29:
> Oh, this might be the input image issue. What kind of image did you subscribe? Was it a color image?

subscribers:
  front_cam:
    # topic_name: '/camera/depth/points'
    topic_name: '/camera/depth/color/points'
    data_type: pointcloud
  color_cam: # for color camera
    topic_name: '/camera/color/image_raw'
    camera_info_topic_name: '/camera/color/camera_info'
    data_type: image
  semantic_cam: # for semantic images
    topic_name: '/front_cam/semantic_image'
    camera_info_topic_name: '/front_cam/camera/depth/camera_info_resized'
    channel_info_topic_name: '/front_cam/channel_info'
    data_type: image
    
    As above, when I use the D435i camera, make sure to input color pictures.

MRZHUGH, 2024-02-29:
[ INFO] [1709173219.924276172]: Subscribed to Image topic: /camera/color/image_raw, Camera info topic: /camera/color/camera_info. Channel info topic: Not found. Using channels: rgb
[ INFO] [1709173219.925903082]: Subscribed to PointCloud2 topic: /camera/depth/color/points
[ INFO] [1709173219.928238252]: Subscribed to Image topic: /front_cam/semantic_image, Camera info topic: /front_cam/camera/depth/camera_info_resized, Channel info topic: /front_cam/channel_info
[ INFO] [1709173220.260316890]: pointcloud_channel_fusion_dict: {'default': 'average', 'rgb': 'color'}
[ INFO] [1709173220.260602523]: image_channel_fusion_dict: {'default': 'exponential', 'feat_.*': 'exponential', 'rgb': 'color'}
Start loading plugins...
Loaded plugins are  min_filter smooth_filter inpainting erosion
[ INFO] [1709173222.165785336]: [ElevationMappingCupy] finish initialization
Layer rgb not found, adding it to the semantic map
[elevation_mapping-1] process has died [pid 256269, exit code -11, cmd 

As above, terminal outputs all information.

mktk1117, 2024-02-29:
Oh so it does happen with the turtlesim example?
Maybe there's some issue in the dependencies.

Have you tried with docker already?

MRZHUGH, 2024-02-29:
> Oh so it does happen with the turtlesim example? Maybe there's some issue in the dependencies.
> 
> Have you tried with docker already?

Thank you for your reply. The main reason is that there is no detailed error message and I don’t know how to fix it. I am currently trying to run it under docker.

MRZHUGH, 2024-02-29:
> Oh so it does happen with the turtlesim example? Maybe there's some issue in the dependencies.
> 
> Have you tried with docker already?

I only positioned the channel input in this function to only have  rgb.

`def update_layers_pointcloud(self, points_all, channels, R, t, elevation_map):`

Furthermore,  I'm not sure if my parameters are misconfigured：

[image]

mktk1117, 2024-03-06:
Thanks for sharing the details.
Could you check what's in the /camera/camera_info' topic?

If it is a color image, you can leave it, or add ["rgb"] as the channel name.