leggedrobotics/elevation_mapping_cupy #87: ERROR: Package 'networkx' requires a different Python: 3.8.10 not in '>=3.9'

MRZHUGH, 2024-02-29:
When I use docker-->./build.sh，have a error: 

`Downloading https://download.pytorch.org/whl/networkx-3.2.1-py3-none-any.whl (1.6 MB)
ERROR: Package 'networkx' requires a different Python: 3.8.10 not in '>=3.9'`

I haven't found a good solution yet.

mktk1117, 2024-02-29:
can you try pulling the image first by just running run.sh?
There maybe some udpate in the networkx library.

MRZHUGH, 2024-03-01:
> can you try pulling the image first by just running run.sh? There maybe some udpate in the networkx library.

Thanks, I will try it .

yyagin, 2024-04-05:
It seems that NetworkX does not support Python 3.8 beyond version 3.2. The latest version that can run with Python 3.8 is 3.1. You can find the link for it [here](https://networkx.org/documentation/stable/release/release_3.1.html).

I solved the issue by adding pip3 install networkx==3.1 just before line 19 in the Dockerfile.

https://github.com/leggedrobotics/elevation_mapping_cupy/blob/ee9a15521bb8d270cb1c2e43c7e89ba71e0e5f3c/docker/Dockerfile.x64#L19

MRZHUGH, 2024-04-05:
您好，我已收到您的邮件，我会尽快给您回复的。祝 工作顺利 生活愉快！

closed 2024-04-24 by mktk1117