leggedrobotics/elevation_mapping_cupy #88: error about

crankler, 2024-03-11:
Hello, thank you for your teams work, it's really helpful.
However, when I run this command ```roslaunch elevation_mapping_cupy elevation_mapping_cupy.launch``` in the terminal, I got this error :
```
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ModuleNotFoundError: No module named 'simple_parsing'

At:
  /home/yc/catkin_map/catkin_eth_cupy/src/elevation_mapping_cupy/elevation_mapping_cupy/script/elevation_mapping_cupy/parameter.py(8): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(848): exec_module
  <frozen importlib._bootstrap>(686): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  /home/yc/catkin_map/catkin_eth_cupy/src/elevation_mapping_cupy/elevation_mapping_cupy/script/elevation_mapping_cupy/__init__.py(1): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(848): exec_module
  <frozen importlib._bootstrap>(686): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap>(961): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load

```
I have install simple_parsing with pip, but the error still exists. I still don't know the reason.
So do you know how to fix it? Thanks you for your reply.

georgflick, 2024-03-11:
You just need to install the package with `pip install simple-parsing` and **not** with `--user`. 
The package should be then here:
`/usr/local/lib/python3.8/dist-packages/simple_parsing`

crankler, 2024-03-11:
Tkank you for your help. However, it still doesn't wrok.
When I input this command:
```yc@yc-NUC11PHi7:~/catkin_map/catkin_eth_cupy$ pip3 install simple_parsing --user
Requirement already satisfied: simple_parsing in /home/yc/.local/lib/python3.11/site-packages (0.1.5)
Requirement already satisfied: docstring-parser~=0.15 in /home/yc/.local/lib/python3.11/site-packages (from simple_parsing) (0.15)
Requirement already satisfied: typing-extensions>=4.5.0 in /home/yc/anaconda3/lib/python3.11/site-packages (from simple_parsing) (4.9.0)
```
and when I launch the file, the error shows here:
```
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ModuleNotFoundError: No module named 'simple_parsing'

At:
  /home/yc/catkin_map/catkin_eth_cupy/src/elevation_mapping_cupy/elevation_mapping_cupy/script/elevation_mapping_cupy/parameter.py(8): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(848): exec_module
  <frozen importlib._bootstrap>(686): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  /home/yc/catkin_map/catkin_eth_cupy/src/elevation_mapping_cupy/elevation_mapping_cupy/script/elevation_mapping_cupy/__init__.py(1): <module>
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap_external>(848): exec_module
  <frozen importlib._bootstrap>(686): _load_unlocked
  <frozen importlib._bootstrap>(975): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load
  <frozen importlib._bootstrap>(219): _call_with_frames_removed
  <frozen importlib._bootstrap>(961): _find_and_load_unlocked
  <frozen importlib._bootstrap>(991): _find_and_load

```
By the way, I use the Ubuntu20.04, cuda 12.4, ros-notetic

georgflick, 2024-03-11:
seems like you installed it with `--user`, so you have to uninstall it first, and reinstall it without the `--user` flag.

`pip uninstall simple-parsing typing-extensions docstring-parser`
and then install it again with `pip install simple-parsing`
should fix the problem.

crankler, 2024-03-11:
Yes, when I use conda with python3.8, I reinstall simple-parsing, and it works.
Thanks again for your quick reply here.

crankler, 2024-03-11:
Sorry again, when I launch ```roslaunch elevation_mapping_cupy turtlesim_semantic_pointcloud_example.launch``` and ````roslaunch elevation_mapping_cupy turtlesim_semantic_image_example.launch```, there also shows the error:
```
Traceback (most recent call last):
  File "/home/yc/catkin_map/catkin_eth/devel/lib/semantic_sensor/image_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yc/catkin_map/catkin_eth/src/elevation_mapping_cupy/sensor_processing/semantic_sensor/script/semantic_sensor/image_node.py", line 21, in <module>
    from semantic_sensor.image_parameters import ImageParameter
  File "/home/yc/catkin_map/catkin_eth/src/elevation_mapping_cupy/sensor_processing/semantic_sensor/script/semantic_sensor/image_parameters.py", line 2, in <module>
    from simple_parsing.helpers import Serializable
ModuleNotFoundError: No module named 'simple_parsing'
[front_cam-5] process has died [pid 191764, exit code 1, cmd /home/yc/catkin_map/catkin_eth/devel/lib/semantic_sensor/image_node.py front_cam_image __name:=front_cam __log:=/home/yc/.ros/log/4915dd76-dfc3-11ee-876d-89b345875324/front_cam-5.log].
log file: /home/yc/.ros/log/4915dd76-dfc3-11ee-876d-89b345875324/front_cam-5*.log

```

and 
```
Traceback (most recent call last):
  File "/home/yc/catkin_map/catkin_eth/devel/lib/semantic_sensor/pointcloud_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yc/catkin_map/catkin_eth/src/elevation_mapping_cupy/sensor_processing/semantic_sensor/script/semantic_sensor/pointcloud_node.py", line 17, in <module>
    from semantic_sensor.pointcloud_parameters import PointcloudParameter
  File "/home/yc/catkin_map/catkin_eth/src/elevation_mapping_cupy/sensor_processing/semantic_sensor/script/semantic_sensor/pointcloud_parameters.py", line 2, in <module>
    from simple_parsing.helpers import Serializable
ModuleNotFoundError: No module named 'simple_parsing'
[front_cam-5] process has died [pid 187311, exit code 1, cmd /home/yc/catkin_map/catkin_eth/devel/lib/semantic_sensor/pointcloud_node.py front_cam __name:=front_cam __log:=/home/yc/.ros/log/72941ca0-dfc1-11ee-876d-89b345875324/front_cam-5.log].
log file: /home/yc/.ros/log/72941ca0-dfc1-11ee-876d-89b345875324/front_cam-5*.log

```

I wonder why this issue can exist, and how can I fix it?
Thanks again here.

crankler, 2024-03-12:
I've solve this issue,
just install the commands:
```
sudo pip3 install simple_parsing
sudo pip3 install numpy==1.21.1
sudo apt-get install ros-noetic-ros-nump
```
I can launch both command in bash.

closed 2024-03-12 by crankler