leggedrobotics/elevation_mapping_cupy #90: [elevation_mapping-5] process has died

CYJ00, 2024-03-26:
My environment:
NVIDIA Jetson AGX Orin Developer Kit
Jetpack: 5.1.3
CUDA: 11.4.315
cuDNN: 8.6.0.166
ROS: noetic

When I run turtlesim_simple_exam.launch, the error below is occurred.
"""
process[gazebo-1]: started with pid [11376]
process[spawn_urdf-2]: started with pid [11381]
process[waffle_state_publisher-3]: started with pid [11382]
process[rviz-4]: started with pid [11383]
process[elevation_mapping-5]: started with pid [11384]
[ INFO] [1711430243.712568343]: Subscribed to Image topic: /camera/rgb/image_raw, Camera info topic: /camera/depth/camera_info. Channel info topic: Not found. Using channels: rgb
[ INFO] [1711430243.719868421]: Subscribed to PointCloud2 topic: /camera/depth/points
[ INFO] [1711430243.735310699]: Subscribed to Image topic: /front_cam/semantic_image, Camera info topic: /front_cam/camera/depth/camera_info_resized, Channel info topic: /front_cam/channel_info
[ INFO] [1711430244.432520981]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1711430244.440608876]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1711430245.075349009]: pointcloud_channel_fusion_dict: {'default': 'average', 'rgb': 'color'}
[ INFO] [1711430245.077350087]: image_channel_fusion_dict: {'default': 'exponential', 'feat_.*': 'exponential', 'rgb': 'color'}
[ INFO] [1711430246.066638018]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1711430246.138029376, 0.009000000]: Physics dynamic reconfigure ready.
[ INFO] [1711430247.324919858, 0.229000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1711430247.348644304, 0.229000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1711430247.749303315, 0.229000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1711430247.753000538, 0.229000000]: Starting Laser Plugin (ns = /)
[ INFO] [1711430247.757280360, 0.229000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1711430247.861724103, 0.229000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1711430247.861909097, 0.229000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1711430247.863501018, 0.229000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1711430247.865162380, 0.229000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1711430247.866073365, 0.229000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1711430247.868995893, 0.229000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1711430247.869721020, 0.229000000]: DiffDrive(ns = //): Advertise odom on odom 
[spawn_urdf-2] process has finished cleanly
log file: /home/youngjoo/.ros/log/f89d5fc0-eb22-11ee-8f5d-48b02decd57a/spawn_urdf-2*.log
Start loading plugins...
Loaded plugins are  min_filter smooth_filter inpainting erosion
[ INFO] [1711430256.196444461, 7.547000000]: [ElevationMappingCupy] finish initialization
Layer rgb not found, adding it to the semantic map
[ INFO] [1711430261.128394719, 12.210000000]: Clearing map with initializer.
[ INFO] [1711430261.149272603, 12.232000000]: Initializing map with points using linear
[ WARN] [1711430261.307094297, 12.349000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint (parent odom) at time 12.292000 according to authority unknown_publisher
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint (parent odom) at time 12.292000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.7/src/buffer_core.cpp
[elevation_mapping-5] process has died [pid 11384, exit code -11, cmd /home/youngjoo/Documents/2levation_ws/devel/lib/elevation_mapping_cupy/elevation_mapping_node __name:=elevation_mapping __log:=/home/youngjoo/.ros/log/f89d5fc0-eb22-11ee-8f5d-48b02decd57a/elevation_mapping-5.log].
log file: /home/youngjoo/.ros/log/f89d5fc0-eb22-11ee-8f5d-48b02decd57a/elevation_mapping-5*.log
^C[rviz-4] killing on exit
[waffle_state_publisher-3] killing on exit
[gazebo-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
"""

If I run elevation_mapping_cupy.launch without setting up the setup.yaml file, I got frame not found error.
"""
process[elevation_mapping-1]: started with pid [31699]
[ INFO] [1711433224.315775825]: Subscribed to Image topic: /camera/rgb/image_raw, Camera info topic: /camera/depth/camera_info. Channel info topic: Not found. Using channels: rgb
[ INFO] [1711433224.322221552]: Subscribed to PointCloud2 topic: /camera/depth/points
[ INFO] [1711433224.336588405]: Subscribed to Image topic: /front_cam/semantic_image, Camera info topic: /front_cam/camera/depth/camera_info_resized, Channel info topic: /front_cam/channel_info
[ INFO] [1711433225.350960292]: pointcloud_channel_fusion_dict: {'default': 'average', 'rgb': 'color'}
[ INFO] [1711433225.351957065]: image_channel_fusion_dict: {'default': 'exponential', 'feat_.*': 'exponential', 'rgb': 'color'}
Start loading plugins...
Loaded plugins are  min_filter smooth_filter inpainting erosion
[ INFO] [1711433231.011774639]: [ElevationMappingCupy] finish initialization
[ERROR] [1711433232.121536898]: "base_footprint" passed to lookupTransform argument source_frame does not exist. 
[ERROR] [1711433233.695018184]: "base_footprint" passed to lookupTransform argument source_frame does not exist. 
[ERROR] [1711433234.778879677]: "base_footprint" passed to lookupTransform argument source_frame does not exist. 
^C[elevation_mapping-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
"""

If I run elevation_mapping_cupy.launch with setting up the setup.yaml file, [elevation_mapping-5] process has died.
"""
process[elevation_mapping-1]: started with pid [32559]
[ INFO] [1711433321.883426655]: Subscribed to Image topic: /camera/rgb/image_raw, Camera info topic: /camera/depth/camera_info. Channel info topic: Not found. Using channels: rgb
[ INFO] [1711433321.892804459]: Subscribed to PointCloud2 topic: /points_concat
[ INFO] [1711433321.902103223]: Subscribed to Image topic: /front_cam/semantic_image, Camera info topic: /front_cam/camera/depth/camera_info_resized, Channel info topic: /front_cam/channel_info
[ INFO] [1711433322.930466625]: pointcloud_channel_fusion_dict: {'default': 'average', 'rgb': 'color'}
[ INFO] [1711433322.931459494]: image_channel_fusion_dict: {'default': 'exponential', 'feat_.*': 'exponential', 'rgb': 'color'}
Start loading plugins...
Loaded plugins are  min_filter smooth_filter inpainting erosion
[ INFO] [1711433327.929947527]: [ElevationMappingCupy] finish initialization
[ INFO] [1711433328.383578604]: Clearing map with initializer.
[ INFO] [1711433328.384056206]: Initializing map with points using linear
[elevation_mapping-1] process has died [pid 32559, exit code -11, cmd /home/youngjoo/Documents/Elevation_ws/devel/lib/elevation_mapping_cupy/elevation_mapping_node __name:=elevation_mapping __log:=/home/youngjoo/.ros/log/de72a092-eb36-11ee-a38f-48b02decd57a/elevation_mapping-1.log].
log file: /home/youngjoo/.ros/log/de72a092-eb36-11ee-a38f-48b02decd57a/elevation_mapping-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
"""

The requirements are like below.
numpy==1.22.4
ruamel.yaml==0.18.6
ruamel.yaml.clib==0.2.8
opencv-python==4.9.0.80
scikit-image==0.19.0
scipy==1.7.0
Shapely==1.7.1
simple-parsing==0.1.5
matplotlib==3.7.5
catkin-tools==0.9.4
cupy-cuda11x==12.3.0

catkin build elevation_mapping_cupy had no error.
It works 2 weeks ago. But suddenly it doesn't work.
Can you give me some advice?

mktk1117, 2024-03-28:
umm, did you pull the recent changes?
The only difference I added is the parameter called `always_clear_with_initializer` which uses `clear_with_initializer` for initializing the map. But the default param is set to false. Can you try checking out the old commit and see if it persist?

CYJ00, 2024-06-17:
After I flashed my Orin and reinstalled the code, it worked. It was probably a dependency issue with other programs I had installed. Thank you!

closed 2024-06-17 by CYJ00