leggedrobotics/elevation_mapping_cupy #99: An uneven map appears when creating a map on a flat surface

nanbwrn, 2024-07-21:
When I used it to build an elevation map for the quadruped robot, I found that even when the robot is on a flat surface, the created map is uneven. This causes the robot to move very unstably. Could you please suggest which parameters should be modified to improve this? Below are my settings for go1_core_param.yaml, go1_example_setup.yaml, and convex_plane_decomposition.yaml. 
Thank you all for your responses.

[image]

[image]

[convex_plane_decomposition.txt](https://github.com/user-attachments/files/16324537/convex_plane_decomposition.txt)
[go1_core_param.txt](https://github.com/user-attachments/files/16324538/go1_core_param.txt)
[go1_example_setup.txt](https://github.com/user-attachments/files/16324539/go1_example_setup.txt)
[plugin_config.txt](https://github.com/user-attachments/files/16324540/plugin_config.txt)

mktk1117, 2024-08-07:
Hi, it might be because of the initialization. 
If you disable the initialization, how it looks like? (you can put false to use_initializer_at_start)

lucarossini-iit, 2024-09-24:
HI @mktk1117, are you speaking about setting use_initializer_at_art = false in the convex_plane_decomposition parameters or the elevation_mapping ones? I am observing that the convex_plane_decomposition creates planes and boundaries also in zones unexplored by the elevation_mapping as it is happing in the case of  @nanbwrn at the back of the robot. Is there a way to disable this or create planes at the same height of the  boundary points of the planes in the elevation map (I mean extend the orange plane on the left of the robot of the previous image in order to cover the zone on the back of the robot)? 

Thank you.

lucarossini-iit, 2024-09-25:
Let me rephrase: is there a way to deactivate the inpaint in the GridMapPreprocessing, or is there a way to avoid drawing a convex region in correspondence of NaN and inpainted regions?

Thank you again!