leggedrobotics/elevation_mapping_cupy #102: Question about Using High-Frequency Leg Odometry and Low-Frequency Outputs for Mapping in Quadruped Robots

zma69650, 2024-08-13:
Thank you! This is a great work! However, I noticed that when using elevation maps to make a quadruped robot walk on complex terrains, it heavily relies on the robot's state estimation. Relying solely on its own perception isn't sufficient, requiring the fusion of visual or lidar information. But in doing so, the output frequency of the state estimation is very low, which isn't sufficient for the controller's use. The paper VILENS: Visual, Inertial, Lidar, and Leg Odometry for All-Terrain Legged Robots discusses using high-frequency leg odometry to satisfy the controller and then using low-frequency outputs for mapping.

Is this approach correct? If so, how should I set it up?

mktk1117, 2024-09-02:
Hi, it makes sense to use higher frequency odometry to update the map. Otherwise you cannot use all pointcloud data with low-frequency odometry. 
Unfortunately I cannot help setting up the VILENS repo, as I never used it, so please ask in that repo for setting that up.

mktk1117, 2024-09-02:
To use the odometry from other source, just change the map_frame parameter to the frame_id of your odometry source.

closed 2024-09-03 by zma69650