leggedrobotics/elevation_mapping_cupy #104: undefined reference to `ffi_closure_alloc@LIBFFI_CLOSURE_7.0'

benijohn, 2024-10-28:
Hi, I have been working on getting this built on a Jetson Orin AGX for the last few days and I am out of ideas. I first had issues with opencv as the version installed with apt did not include the xphoto directory.. after some questionable work, I rebuilt opencv from scratch and copied the files into the /usr/include path and finally got past that issue. However, I have been getting this error all along as well. I originally thought they might be related(and they might still be?) but I am not sure what try next.

/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_closure_alloc@LIBFFI_CLOSURE_7.0'
/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_prep_closure_loc@LIBFFI_CLOSURE_7.0'
/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_type_uint8@LIBFFI_BASE_7.0'
/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_prep_cif@LIBFFI_BASE_7.0'
/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_type_pointer@LIBFFI_BASE_7.0'
/usr/bin/ld: /lib/aarch64-linux-gnu/libp11-kit.so.0: undefined reference to `ffi_type_uint64@LIBFFI_BASE_7.0'
collect2: error: ld returned 1 exit status

I reinstalled both libffi and libp11-kit. I tried catkin_make clean. I have also checked they are linked by running:

$ ldd /lib/aarch64-linux-gnu/libp11-kit.so.0
	linux-vdso.so.1 (0x0000ffff95a4e000)
	/usr/lib/aarch64-linux-gnu/libffi.so.7 (0x0000ffff958b9000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff9585b000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff9582a000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff956b7000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff95a1e000)

The paths appear correct to me. 

Any ideas on what to try next? Has anyone seen this issue before? It seems most things online point to a missing ffi library, but that is not the case here.

Gatswei, 2024-11-09:
I encounter the same problem. Have you solved it?

benijohn, 2024-11-14:
Not yet. I am reflashing my Jetson with a clean JetPack install and will try again later today. I will update here if I have any luck. I have built this successfully many times on x64 machines so this seems to be Jetson related. Are you also on Jetson?

benijohn, 2025-01-15:
Just wanted to update here. I did finally get somewhere with the above error. It seems to be related to using a Conda environment. I had previously only seen this on the Jetson platform, but when build on another x64 machine I started looking into a bit more. I ended up creating an environment using venv instead and this seems to do much better. I think the issue is with LIBFFI7 being linked to LIBFFI8 in the conda environment. There may be other ways around this, but using venv has worked for me on x64 and jetson. On the jetson, I have only built the core package and am not currently using plane decomp or semantic sensors. 

Below are the steps I used:

#create virtual environment python 3.8
python3.8 -m venv nameOfEnvironment
#upgrade pip setup tools
python -m pip install -U setuptools pip
cd elevation_mapping_cupy
pip install -r requirements.txt
pip install rospkg empy==3.3.4

export TORCH_INSTALL=https://developer.download.nvidia.cn/compute/redist/jp/v511/pytorch/torch-2.0.0+nv23.05-cp38-cp38-linux_aarch64.whl
pip install Cython
###### I did not install numpy 1.24 it conflicts with other requirements #######
### python -m pip install numpy==’1.24.1’
python -m pip install --no-cache $TORCH_INSTALL

pip install cupy-cuda11x

sudo apt install ros-noetic-pybind11-catkin
sudo apt install ros-noetic-grid-map-core ros-noetic-grid-map-msgs ros-noetic-grid-map-ros

sudo apt install gfortran

sudo apt install libeigen3-dev

### I skipped the libopencv install due to previous issues. I will try again at a later point

sudo apt install libgmp-dev
sudo apt install libmpfr-dev
sudo apt install libboost-all-dev

I did install torchvision, but detectron still fails saying torch no installed. but it is installed. 

git clone --branch release/0.15 https://github.com/pytorch/vision torchvision
cd torchvision/
export BUILD_VERSION=0.15.1
python3 setup.py install --user

catkin build elevation_mapping_cupy

complikator, 2025-03-04:
Hi,
I've encountered similar problems with opencv at an earlier stage; any suggestions on how to solve it?
I've tried purging it according to some suggestions of dustynv, but it gets me into other problems

MyDearestJingo, 2025-03-13:
I met the same problem with miniconda, and solved it by `rm ${CONDA_PREFIX}/lib/libffi.7.so ${CONDA_PREFIX}/lib/libffi.so.7`. They conficted to the system lib