leggedrobotics/elevation_mapping_cupy #109: semantic image layers don't work with camera resolutions other than 640x320

shin-ka, 2025-01-27:
Hi,

In my project (in simulation) I am trying to generate semantic layers, similar to this launch file : turtlesim_semantic_image_example.launch. Basically the semantic node is segmenting a particular object in the scene and I am adding a semantic layer to the /elevation_map_raw topic which contains info about where this object is lying on the grid.

I am able to get correct correspondences between image and grid and the output semantic layer looks correct for image resolution 640x320. However, if I change the resolution to any other value, the object location shifts and eventually the semantic layer has no information of the object anymore.

I selected the resolution of 640x320 in my project based on the turtlebot example where the turtlebot has camera resolution of 640x320.

Could you please help me out what might be causing this?

shin-ka, 2025-02-05:
Hi, yes the camera info topic is always updated with every new camera resolution I test with.

shin-ka, 2025-02-05:
When I try increasing the resolution, it appears as if the information in the semantic grid layer shifts towards the left and at 1280x720 resolution the grid is completely empty.
Looks like the projection of the image on the grid doesn't happen anymore.

Could you please point out which parts of the code I should look into to find out the error?

mktk1117, 2025-02-05:
Could you check the Subscriber params?

For example these camera_info topic params.
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/00bbe01e3b6041f9f3eb86997637becd725a5291/elevation_mapping_cupy/config/setups/turtle_bot/turtle_bot_semantics_image.yaml#L18

You can do `rosparam list | grep camera_info_topic_name`
to listup the related parameters and check the values with `rosparam get <param_name>`

shin-ka, 2025-02-09:
Hi, was able to solve the issue. There is a resize param that was set to 0.5 which was resizing the images. Changed it to 1.0 and it works with other resolutions.

Thanks for your support!

closed 2025-02-09 by shin-ka