leggedrobotics/elevation_mapping_cupy #111: Map stop mapping and start cleaning everything

GLMDaleth, 2025-04-02:
Hello there,

I'm actually using your package which is awesome for mapping rough terrain, 

But i'm facing some probleme using it, actually when i launch the node the mapping starts well and everything is find but after moving a bit with my robot the mapping stop working and instead the map start getting cleaned.

I have no idea where this come from, do you know what could happens ? 

If so which parameters i should change and why? 

regards.

mktk1117, 2025-04-05:
The map is made around the `base_frame` you set in the rosparam.
Maybe your robot has different frame_id than this, so the map is not moving with your robot. 

base_frame: 'base_footprint'                              # The robot's base frame. This frame will be a center of the map.

GLMDaleth, 2025-04-07:
hi @mktk1117, 
thanks for your answer, all the frame id and topics are well set, the map is moving with the robot centered in it and everything works fine but after some times we found that the mapping isn't working anymore and so the map start disapearing.

We don't know why and even changing some parameters didn't do anything.

any idea ? 

regards.

mktk1117, 2025-04-07:
Do you have any screen recording?

GLMDaleth, 2025-04-07:
here is a video of whats happening

[image]

mktk1117, 2025-04-07:
umm, it's weird...
Can you double check if these parameters make sense for your setup? (ramped_height_range_a, b, c)
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/6ba8383978e8437752f9cea0d15ce6446057c623/elevation_mapping_cupy/config/core/core_param.yaml#L23

Also can you disable these and see what happens?
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/6ba8383978e8437752f9cea0d15ce6446057c623/elevation_mapping_cupy/config/core/core_param.yaml#L51
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/6ba8383978e8437752f9cea0d15ce6446057c623/elevation_mapping_cupy/config/core/core_param.yaml#L49

GLMDaleth, 2025-04-07:
i've tried disabling the 2 parameters and nothing changed.
For the "ramped_height_range_*" parameters i don't exactly understand what it refers to and so i don't know if there make sense for my setup, do you have a better explanation of it ?

GLMDaleth, 2025-11-25:
@iiisaac40 Hello,
So to be honest we didn't manage to find the real issue, but we changed the robot that was using elevation mapping for a robot with a bigger computer and the problem disappeared.

wrj700, 2026-01-11:
> [@iiisaac40](https://github.com/iiisaac40) Hello, So to be honest we didn't manage to find the real issue, but we changed the robot that was using elevation mapping for a robot with a bigger computer and the problem disappeared.

@GLMDaleth Hello, may I ask if you have encountered the problem of point clouds being blocked by your body? How can you obtain this part of the elevation map?