leggedrobotics/elevation_mapping_cupy #115: cleaning up points from robot's body

anahrendra, 2025-07-05:
Hi! Thanks for sharing your great work.
I have a question regarding how to set some parameters to handle unreliable points (especially those coming from the robot's body itself as shown in the fig i attached below).

Is there any params or best practice to mitigate this kind of issue? Thanks in advance!

[image]

mktk1117, 2025-07-07:
The best way is to use self filter or something like that which can filter the points from the robot body.
https://wiki.ros.org/robot_self_filter

But you can also tune this parameter for quick fix.
https://github.com/leggedrobotics/elevation_mapping_cupy/blob/20a8a26b67a995b43eb44c23568854d1fed82a52/elevation_mapping_cupy/config/core/core_param.yaml#L21

anahrendra, 2025-07-08:
@mktk1117 thanks for  your prompt reply! will try those fixes soon!

closed 2025-07-08 by anahrendra